edison-fw/meta-intel-edison #128: ACPI table is not loaded

shawnhsoia97, 2021-03-26:
Hi , I encounter some problem for ACPI
My ACPI table not loaded by initramfs.
I have three ACPI files follow below:
```
ACPI_TABLES ?= "spidev.asl mcu_thales.asl i2c3028.asl "
#ACPI_TABLES += " mcu_5g.asl i2c3028.asl "
ACPI_FEATURES_edison ?= "uart_2w i2c spi"

```
But the `dmesg`  information is not show "load xxx.aml",which show the message below only:
```
root:~# dmesg | grep acpi
[    0.000000] Linux version 5.6.0-edison-acpi-standard (oe-user@oe-host) (gcc version 9.2.0 (GCC)) #1 SMP Tue Mar 2 01:46:52 UTC 2021
[    0.035660] ACPI: LAPIC (acpi_id[0x00] lapic_id[0x00] enabled)
[    0.035671] ACPI: LAPIC (acpi_id[0x02] lapic_id[0x02] enabled)
[    0.177150] acpi PNP0A08:00: _OSC: OS supports [ExtendedConfig ASPM ClockPM Segments MSI HPX-Type3]
[    0.177789] acpi PNP0A08:00: _OSC: OS now controls [PME AER PCIeCapability LTR]
[    0.177800] acpi PNP0A08:00: FADT indicates ASPM is unsupported, using BIOS configuration
[    0.177888] acpi PNP0A08:00: [Firmware Info]: MMCONFIG for domain 0000 [bus 00-00] only partially covers this bridge
[    4.640993] usb usb1: Manufacturer: Linux 5.6.0-edison-acpi-standard xhci-hcd
[    4.644905] usb usb2: Manufacturer: Linux 5.6.0-edison-acpi-standard xhci-hcd
```

I currently use branch zeus with version "U-Boot 2020.10 " ,But  I used branch zeus with version "U-Boot 2020.04 " before use the this version , I have not this issues and this ACPI is correctly.

Originally I used acpi to make rtc1 load faster than rtc0 ,but now rtc1 loads slower than rtc0

```
root:~# dmesg | grep rtc
[    3.050660] rtc_cmos 00:00: registered as rtc0
[    3.050883] rtc_cmos 00:00: no alarms, 242 bytes nvram
[    3.096086] hctosys: unable to open rtc device (rtc1)
```

In addition I can found ACPI table in system, but I think it not loaded.

```
root:/home/bovia/daemon# ls /sys/kernel/config/acpi/table/
i2c3028  mcu_thales  spidev

```

I don't know why  this problem will happen, How about to solve it? Thank you!

htot, 2021-03-26:
Can you do `dmesg | grep ACPI`? I get:
```
ferry@ferry-quad:~/Downloads$ grep ACPI "dmesg 5.10-rc2.txt.txt" 
[       0.001619] ACPI: Early table checksum verification disabled
[       0.001637] ACPI: RSDP 0x00000000000E4500 000024 (v02 U-BOOT)
[       0.001669] ACPI: XSDT 0x00000000000E45E0 00004C (v01 U-BOOT U-BOOTBL 20201011 INTL 00000000)
[       0.001710] ACPI: FACP 0x00000000000E5330 0000F4 (v06 U-BOOT U-BOOTBL 20201011 INTL 00000000)
[       0.001754] ACPI: DSDT 0x00000000000E4780 000AA4 (v02 U-BOOT U-BOOTBL 00010000 INTL 20190215)
[       0.001789] ACPI: MCFG 0x00000000000E5454 00003C (v01 U-BOOT U-BOOTBL 20201011 INTL 00000000)
[       0.001821] ACPI: APIC 0x00000000000E5490 000048 (v02 U-BOOT U-BOOTBL 20201011 INTL 00000000)
[       0.001852] ACPI: CSRT 0x00000000000E54E0 000058 (v00 U-BOOT U-BOOTBL 20201011 INTL 00000000)
[       0.001883] ACPI: SPCR 0x00000000000E5540 000050 (v02 U-BOOT U-BOOTBL 20201011 INTL 00000000)
[       0.001941] ACPI: Local APIC address 0xfee00000
[       0.014776] ACPI: Local APIC address 0xfee00000
[       0.014851] Using ACPI (MADT) for SMP configuration information
[       0.014870] ACPI: SPCR: console: uart,mmio,0xff010180
[       0.078762] ACPI: Core revision 20200925
[       0.099618] ACPI FADT declares the system doesn't support PCIe ASPM, so disable it
[       0.099631] ACPI: bus type PCI registered
[       0.122242] ACPI: Added _OSI(Module Device)
[       0.122255] ACPI: Added _OSI(Processor Device)
[       0.122264] ACPI: Added _OSI(3.0 _SCP Extensions)
[       0.122273] ACPI: Added _OSI(Processor Aggregator Device)
[       0.122285] ACPI: Added _OSI(Linux-Dell-Video)
[       0.122295] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[       0.122305] ACPI: Added _OSI(Linux-HPI-Hybrid-Graphics)
[       0.127700] ACPI: 1 ACPI AML tables successfully acquired and loaded
[       0.129847] ACPI: Interpreter enabled
[       0.129915] ACPI: (supports S0)
[       0.129926] ACPI: Using IOAPIC for interrupt routing
[       0.130026] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[       0.143773] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[       0.174427] ACPI: bus type USB registered
[       1.310615] pnp: PnP ACPI init
[       1.310967] pnp 00:00: Plug and Play ACPI device, IDs PNP0b00 (active)
[       1.311579] system 00:01: Plug and Play ACPI device, IDs PNP0c02 (active)
[       1.313084] pnp: PnP ACPI: found 2 devices
[       2.558614] 8250_mid 0000:00:04.1: using ACPI for GPIO lookup
[       2.594403] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       2.594476] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       2.598311] rtc_cmos 00:00: using ACPI for GPIO lookup
[       2.602304] i2c-designware-pci 0000:00:08.0: using ACPI for GPIO lookup
[       2.608680] i2c-designware-pci 0000:00:09.1: using ACPI for GPIO lookup
[       2.627368] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       2.627441] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       2.633971] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       2.634045] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       2.640044] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       2.640163] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       3.453110] ACPI: Host-directed Dynamic ACPI Table Load:
[       3.453155] ACPI: SSDT 0xFFFF8DF04726B180 0000A6 (v05     SPIDEV   00000001 INTL 20190215)
[       3.556875] ACPI: Host-directed Dynamic ACPI Table Load:
[       3.556920] ACPI: SSDT 0xFFFF8DF047213600 000193 (v05     LEDS    00000001 INTL 20190215)
[       3.613604] ACPI: Host-directed Dynamic ACPI Table Load:
[       3.613670] ACPI: SSDT 0xFFFF8DF047232000 000DB9 (v05     ARDUINO  00000001 INTL 20190215)
[       3.668960] pca953x i2c-INT3491:00: using ACPI for GPIO lookup
[       3.678706] pca953x i2c-INT3491:01: using ACPI for GPIO lookup
[       3.687570] pca953x i2c-INT3491:02: using ACPI for GPIO lookup
[       3.693665] pca953x i2c-INT3491:03: using ACPI for GPIO lookup
[       3.715436] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       3.715511] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       3.738315] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
[       3.738388] tusb1210 dwc3.0.auto.ulpi: using ACPI for GPIO lookup
```

shawnhsoia97, 2021-03-29:
Sorry it's my fault.
```
# dmesg | grep ACPI
[    0.001676] ACPI: Early table checksum verification disabled
[    0.001692] ACPI: RSDP 0x00000000000E4500 000024 (v02 U-BOOT)
[    0.001713] ACPI: XSDT 0x00000000000E45E0 00004C (v01 U-BOOT U-BOOTBL 20210302 INTL 00000000)
[    0.001745] ACPI: FACP 0x00000000000E5330 0000F4 (v06 U-BOOT U-BOOTBL 20210302 INTL 00000000)
[    0.001781] ACPI: DSDT 0x00000000000E4780 000AA4 (v02 U-BOOT U-BOOTBL 00010000 INTL 20190215)
[    0.001805] ACPI: MCFG 0x00000000000E5454 00003C (v01 U-BOOT U-BOOTBL 20210302 INTL 00000000)
[    0.001828] ACPI: APIC 0x00000000000E5490 000048 (v02 U-BOOT U-BOOTBL 20210302 INTL 00000000)
[    0.001851] ACPI: CSRT 0x00000000000E54E0 000058 (v00 U-BOOT U-BOOTBL 20210302 INTL 00000000)
[    0.001873] ACPI: SPCR 0x00000000000E5540 000050 (v02 U-BOOT U-BOOTBL 20210302 INTL 00000000)
[    0.001923] ACPI: Local APIC address 0xfee00000
[    0.035618] ACPI: no legacy devices present
[    0.035624] ACPI: probing for VGA not safe
[    0.035638] ACPI: Local APIC address 0xfee00000
[    0.035660] ACPI: LAPIC (acpi_id[0x00] lapic_id[0x00] enabled)
[    0.035671] ACPI: LAPIC (acpi_id[0x02] lapic_id[0x02] enabled)
[    0.035694] ACPI: IOAPIC (id[0x02] address[0xfec00000] gsi_base[0])
[    0.035739] Using ACPI (MADT) for SMP configuration information
[    0.035753] ACPI: SPCR: console: uart,mmio,0xff010180
[    0.098414] ACPI: Core revision 20200110
[    0.121964] ACPI FADT declares the system doesn't support PCIe ASPM, so disable it
[    0.121973] ACPI: bus type PCI registered
[    0.155308] ACPI: Added _OSI(Module Device)
[    0.155316] ACPI: Added _OSI(Processor Device)
[    0.155322] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.155328] ACPI: Added _OSI(Processor Aggregator Device)
[    0.155337] ACPI: Added _OSI(Linux-Dell-Video)
[    0.155345] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[    0.155352] ACPI: Added _OSI(Linux-HPI-Hybrid-Graphics)
[    0.160936] ACPI: 1 ACPI AML tables successfully acquired and loaded
[    0.163084] ACPI: Interpreter enabled
[    0.163146] ACPI: (supports S0)
[    0.163154] ACPI: Using IOAPIC for interrupt routing
[    0.163249] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    0.177110] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[    0.208281] ACPI: bus type USB registered
[    1.258128] pnp: PnP ACPI init
[    1.258466] pnp 00:00: Plug and Play ACPI device, IDs PNP0b00 (active)
[    1.259061] system 00:01: Plug and Play ACPI device, IDs PNP0c02 (active)
[    1.260503] pnp: PnP ACPI: found 2 devices
[    4.122247] ACPI: Host-directed Dynamic ACPI Table Load:
[    4.122289] ACPI: SSDT 0xFFFFA1E37C847800 0005D7 (v05        openhub  00000001 INTL 20190215)
[    4.223001] ACPI: Host-directed Dynamic ACPI Table Load:
[    4.223034] ACPI: SSDT 0xFFFFA1E346430E40 0000A6 (v05        SPIDEV   00000001 INTL 20190215)
[    4.332211] ACPI: Host-directed Dynamic ACPI Table Load:
[    4.332246] ACPI: SSDT 0xFFFFA1E346440900 0000E5 (v05        i2c      00000001 INTL 20190215)
```
But why is loading speed slow? Can let it be make faster?

I need to load faster than hctosys driver , because rtc is linked to rtc1,otherwise hctosys will failed.(`[    3.096086] hctosys: unable to open rtc device (rtc1)`)

```
~# dmesg | grep rtc
[    3.053215] rtc_cmos 00:00: registered as rtc0
[    3.053347] rtc_cmos 00:00: no alarms, 242 bytes nvram
[    3.098526] hctosys: unable to open rtc device (rtc1)
[    4.292096] rtc-rv3028 i2c-PRP0001:01: Voltage low, data loss detected.
[    4.292930] rtc-rv3028 i2c-PRP0001:01: Voltage low, data is invalid.
[    4.293262] rtc-rv3028 i2c-PRP0001:01: registered as rtc1

```

htot, 2021-03-29:
I think you mean earlier, not faster? All I can find is: https://bugzilla.kernel.org/show_bug.cgi?id=43010
i2c and spi only get loaded after acpi table is loaded. You can get the acpi tables loaded earlier by putting them in a cpio. But the bugzilla suggestion is to load hctosys later.

shawnhsoia97, 2021-03-30:
Yes,  I mean earlier. Okay, I will try to use acpi-tables.cpio follow [1.3-ACPI-or-not.html#acpi-tables](https://edison-fw.github.io/meta-intel-edison/1.3-ACPI-or-not.html#acpi-tables)

Perform the follow stepping :

- bitbake -c cleansstate acpi-tables
- bitbake acpi-tables
- make edison-image
```
 $ ls out/current/build/tmp/deploy/images/edison/acpi-tables.cpio
out/current/build/tmp/deploy/images/edison/acpi-tables.cpio
```

- Preloading by U-Boot
- Modify `meta-intel-edison/meta-intel-edison-bsp/recipes-bsp/u-boot/files/edison.env`
- `load_edsboot=load mmc 1 ${loadaddr} /boot/bzImage;load mmc 1 ${rdaddr} acpi-tables.cpio;load mmc 1 ${rdaddr} /boot/initrd` instead  of  `load_edsboot=load mmc 1 ${loadaddr} /boot/bzImage;load mmc 1 ${rdaddr} /boot/initrd`

I have performed above step , but the ACPI table was not loaded earlier.
Maybe I am missing some steps? because original `load_edsboot` is `'load mmc 0:9 0x100000 vmlinuz.efi; load mmc 0:9 0x3000000 acpi-tables.cpio'` in  [1.3-ACPI-or-not.html#acpi-tables](https://edison-fw.github.io/meta-intel-edison/1.3-ACPI-or-not.html#acpi-tables)
```
Starting kernel ...

[    2.909372] Initramfs unpacking failed: invalid magic at start of compressed archive
Scanning for Btrfs filesystems
Starting version 247.4+
Kernel with acpi enabled detected
Loading acpi tables
gpioset: at least one GPIO line offset to value mapping must be specified
gpioset: at least one GPIO line offset to value mapping must be specified
Waiting for root device /dev/mmcblk0p8
Found device '/run/media/mmcblk0p8'
Init found, booting...
mount: can't find /boot in /etc/fstab
mount: mounting /boot on /realroot/boot failed: Invalid argument
mount: can't find /lib/modules in /etc/fstab
```
```
~# dmesg | grep ACPI
[    0.002203] ACPI: Early table checksum verification disabled
[    0.002220] ACPI: RSDP 0x00000000000E4500 000024 (v02 U-BOOT)
[    0.002241] ACPI: XSDT 0x00000000000E45E0 00004C (v01 U-BOOT U-BOOTBL 20210330 INTL 00000000)
[    0.002273] ACPI: FACP 0x00000000000E5330 0000F4 (v06 U-BOOT U-BOOTBL 20210330 INTL 00000000)
[    0.002309] ACPI: DSDT 0x00000000000E4780 000AA4 (v02 U-BOOT U-BOOTBL 00010000 INTL 20190215)
[    0.002333] ACPI: MCFG 0x00000000000E5454 00003C (v01 U-BOOT U-BOOTBL 20210330 INTL 00000000)
[    0.002356] ACPI: APIC 0x00000000000E5490 000048 (v02 U-BOOT U-BOOTBL 20210330 INTL 00000000)
[    0.002379] ACPI: CSRT 0x00000000000E54E0 000058 (v00 U-BOOT U-BOOTBL 20210330 INTL 00000000)
[    0.002401] ACPI: SPCR 0x00000000000E5540 000050 (v02 U-BOOT U-BOOTBL 20210330 INTL 00000000)
[    0.002451] ACPI: Local APIC address 0xfee00000
[    0.036147] ACPI: no legacy devices present
[    0.036152] ACPI: probing for VGA not safe
[    0.036167] ACPI: Local APIC address 0xfee00000
[    0.036188] ACPI: LAPIC (acpi_id[0x00] lapic_id[0x00] enabled)
[    0.036200] ACPI: LAPIC (acpi_id[0x02] lapic_id[0x02] enabled)
[    0.036222] ACPI: IOAPIC (id[0x02] address[0xfec00000] gsi_base[0])
[    0.036269] Using ACPI (MADT) for SMP configuration information
[    0.036282] ACPI: SPCR: console: uart,mmio,0xff010180
[    0.098927] ACPI: Core revision 20200110
[    0.122351] ACPI FADT declares the system doesn't support PCIe ASPM, so disable it
[    0.122351] ACPI: bus type PCI registered
[    0.155814] ACPI: Added _OSI(Module Device)
[    0.155821] ACPI: Added _OSI(Processor Device)
[    0.155828] ACPI: Added _OSI(3.0 _SCP Extensions)
[    0.155834] ACPI: Added _OSI(Processor Aggregator Device)
[    0.155843] ACPI: Added _OSI(Linux-Dell-Video)
[    0.155851] ACPI: Added _OSI(Linux-Lenovo-NV-HDMI-Audio)
[    0.155859] ACPI: Added _OSI(Linux-HPI-Hybrid-Graphics)
[    0.161529] ACPI: 1 ACPI AML tables successfully acquired and loaded
[    0.163614] ACPI: Interpreter enabled
[    0.163674] ACPI: (supports S0)
[    0.163682] ACPI: Using IOAPIC for interrupt routing
[    0.163778] PCI: Using host bridge windows from ACPI; if necessary, use "pci=nocrs" and report a bug
[    0.177656] ACPI: PCI Root Bridge [PCI0] (domain 0000 [bus 00-ff])
[    0.208515] ACPI: bus type USB registered
[    1.258617] pnp: PnP ACPI init
[    1.258947] pnp 00:00: Plug and Play ACPI device, IDs PNP0b00 (active)
[    1.259554] system 00:01: Plug and Play ACPI device, IDs PNP0c02 (active)
[    1.260985] pnp: PnP ACPI: found 2 devices
[    3.953101] ACPI: Host-directed Dynamic ACPI Table Load:
[    3.953141] ACPI: SSDT 0xFFFF96B8071D6800 0005D7 (v05        openhub  00000001 INTL 20190215)
[    3.991420] ACPI: Host-directed Dynamic ACPI Table Load:
[    3.991452] ACPI: SSDT 0xFFFF96B8071F3780 0000A6 (v05        SPIDEV   00000001 INTL 20190215)
[    4.024843] ACPI: Host-directed Dynamic ACPI Table Load:
[    4.024876] ACPI: SSDT 0xFFFF96B80710FD00 0000E5 (v05        i2c      00000001 INTL 20190215)

```

```
# dmesg | grep rtc
[    3.035957] rtc_cmos 00:00: registered as rtc0
[    3.036089] rtc_cmos 00:00: no alarms, 242 bytes nvram
[    3.081536] hctosys: unable to open rtc device (rtc1)
[    4.030829] rtc-rv3028 i2c-PRP0001:01: Voltage low, data loss detected.
[    4.031555] rtc-rv3028 i2c-PRP0001:01: Voltage low, data is invalid.
[    4.031887] rtc-rv3028 i2c-PRP0001:01: registered as rtc1

```

htot, 2021-03-30:
If you want to see if the acpi tables are loaded earlier, go into the rescue image (`run do_rescue`). Normally tables are loaded from the file system (so after mounting), but when loaded by U-Boot they are loaded before.

But I wouldn't be surprised if it doesn't help you. I would bet on making hctosys a module and loading that later.

shawnhsoia97, 2021-03-30:
I also tried to making hctosys and loading that later.

But it's a little hard for me that making hctosys a module or find out who performing hctosys.

The munuconfig don't let me select to \<M\> at CONFIG_RTC_HCTOSYS and  CONFIG_RTC_CLASS
```
--- Real Time Clock                                                                                   
  │ │                     [*]   Set system time from RTC on startup and resume                                
  │ │                     (rtc1)  RTC used to set the system time   
```
And also not found these in system

```
 # modinfo hctosys
modinfo: ERROR: Module hctosys not found.
# modinfo rtc
modinfo: ERROR: Module rtc not found.
```
```
~# ls /lib/modules/5.6.0-edison-acpi-standard/kernel/drivers/
bluetooth/ dma/       extcon/    fs/        gnss/      hid/       hwmon/     i2c/       iio/       input/     leds/      media/     mfd/       mmc/       net/       phy/       platform/  pwm/       spi/       thermal/   usb/

```

shawnhsoia97, 2021-03-31:
Thank your help!
I have found the this patch [support hctosys from modular RTC drivers](https://patchwork.ozlabs.org/project/rtc-linux/patch/20191106194625.116692-1-smuckle@google.com/)

It is help for me.
It seem that the problem has been solved in the kernel 5.7 and above.

htot, 2021-03-31:
So I you try the Dunfell image now in PR #126 which has 5.10.24 LTS kernel this patch should be already present.