edison-fw/meta-intel-edison #130: How to use PWM mode?

shawnhsoia97, 2021-05-17:
Hi @htot ,
I want to set PWM mode.
I tried used below code in my .asl files and use /sys/class/pwm class test , but which is not effect.

```
External (_SB_.PCI0.PWM0, DeviceObj)

Scope (\_SB.PCI0.PWM0)
{
    Name (RBUF, ResourceTemplate ()
    {
        PinFunction(Exclusive, PullUp, 0x0001,
            "\\_SB.FLIS", 0, ResourceConsumer, , ) { 12 }
    })
    Method (_CRS, 0, NotSerialized)
    {
        Return (RBUF)
    }
}
```
Please, Could you tell me how to use PWM? I need to switch GP12 to PWM mode1.
By the way I use in branch zeus ,and that not use Arduino development board.
Thank you.

htot, 2021-05-17:
What do you mean zeus does not use Arduino board?

We have this https://github.com/edison-fw/meta-acpi/blob/a48a1c514978038ba17c115c94845341ffb91d54/recipes-bsp/acpi-tables/samples/edison/arduino.asli#L1131

```
Scope (\_SB.PCI0.PWM0)
{
    Name (RBUF, ResourceTemplate ()
    {
        PinFunction(Exclusive, PullUp, 0x0001,
            "\\_SB.FLIS", 0, ResourceConsumer, , ) { 144 }
        PinFunction(Exclusive, PullUp, 0x0001,
            "\\_SB.FLIS", 0, ResourceConsumer, , ) { 145 }
        PinFunction(Exclusive, PullUp, 0x0001,
            "\\_SB.FLIS", 0, ResourceConsumer, , ) { 132 }
        PinFunction(Exclusive, PullUp, 0x0001,
            "\\_SB.FLIS", 0, ResourceConsumer, , ) { 133 }
    })
    Method (_CRS, 0, NotSerialized)
    {
        Return (RBUF)
    }
}
```

shawnhsoia97, 2021-05-20:
> What do you mean zeus does not use Arduino board?
> 

My mean is only use GPIO of Edison , and I did not develop on Arduino board.

> We have this https://github.com/edison-fw/meta-acpi/blob/a48a1c514978038ba17c115c94845341ffb91d54/recipes-bsp/acpi-tables/samples/edison/arduino.asli#L1131
>
> ```
> Scope (\_SB.PCI0.PWM0)
> {
>     Name [...]
Yes , I did follow this example , but I use control pwm based on pwm class (e.g. /sys/class/pwm/pwmchip0/pwm0) that is no  PWM signal created on pwm0(GP12).

htot, 2021-05-21:
I see, it appears you are right. The pin mode needs to be set and there is no infrastructure for that yet. For now there is [hack floating around](https://lore.kernel.org/lkml/20210521135516.64727-1-andriy.shevchenko@linux.intel.com/T/#u) that you might try. It sets the pin modes when the pwm driver is probed, maybe it works for you?

shawnhsoia97, 2021-05-26:
Thank your help ! I appreciate  that.
This works for me. The pwm driver has been correctly probed.

closed 2021-05-26 by shawnhsoia97

andy-shev, 2021-10-03:
Also possible to do this via @xlla patches as pointed out here https://github.com/edison-fw/meta-intel-edison/issues/100.