edison-fw/meta-intel-edison #132: requirement for sudo to use flashall.sh

humanoid2050, 2021-09-02:
The documentation states in a few places that root permissions via sudo are required for flashall.sh on Ubuntu.

Creating the file `/etc/udev/rules.d/46-edison-dfu.rules` with the following line
```
SUBSYSTEM=="usb", ATTRS{idVendor}=="8087", ATTRS{idProduct}=="0a99", MODE="0664", GROUP="plugdev"
```
will allow a user (who is a member of the plugdev group) to use the script without sudo.

htot, 2021-09-03:
That  would be useful to add to the documentation. Would you create a patch to document how to do that and correct current refs to sudo?

humanoid2050, 2021-09-03:
My Markdown-fu is weak, but I will certainly take a crack at it. I think Setting Up > Prerequisites for building is probably the right place to document the fix, and then add references back to that in the building and installing pages that actually use the script. Let me know if you have a better idea for the layout.

humanoid2050, 2021-09-04:
PR #133 has changes for this.

htot, 2021-09-07:
This will appear in gatesgarth soon to be released, thanks.

closed 2021-09-07 by htot