edison-fw/meta-intel-edison #134: `make debian` broken on master and dunfell

humanoid2050, 2021-09-05:
The build scripts do not actually copy the updated image to the toFlash directory.

htot, 2021-09-06:
True. Because I didn't want to pollute the toFlash directory while generating the btrfs subvolume. But probably as long as we support also ext4 it makes sense to behave as the original.

BTW: Have you tried the btrfs image? It allows you to dual boot Yocto/Debian.

htot, 2021-09-07:
https://github.com/htot/meta-intel-edison

closed 2021-09-07 by htot

htot, 2021-09-08:
> If anything, I'd actually like to set up my Edison to be completely based on the SD card, and just flash images to a card the way I do with the other 15 or so embedded computers in my collection.

You can already do that now, as U-Boot can read the SD card to load the kernel.

> Has any thought been given to being able to support btrfs/ext4, emmc/uSD, separate boot/home, yocto/debian and various combinations of those options in a robust way?

Yes, this is already the case. But maybe not all combinations have been tested rigidly.