edison-fw/meta-intel-edison #136: SD card not detected properly in U-Boot

humanoid2050, 2021-10-14:
At least as far back as the Dunfell branch, U-Boot (as installed) does not properly detect an SD card when using a Sparkfun microSD Block. [Product page](https://www.sparkfun.com/products/retired/13041) [Schematic](https://cdn.sparkfun.com/datasheets/Dev/Edison/Micro_SD_Block.pdf) A very long time ago (like when Intel was still supporting the Edison) U-Boot was able to see the uSD card.

When the card slot is empty
```
=> mmc dev 1
Card did not respond to voltage select! : -110
```
When the card slot is populated
```
=> mmc dev 1
MMC: no card present
```

Presumably, this is the U-Boot analog of #135, which focused on the kernel. Investigation into that issue indicated that the Sparkfun board and the Edison-Arduino board use different hardware for the uSD socket. The Sparkfun board has a "normally closed" switch on the Card Detect (CD) line. This means that when no card is present, the CD line is shorted to ground. Inserting the card breaks the contact. The Edison-Arduino board uses a "normally open" switch internally with an external pullup to 1.8V on the line.

I had assumed that the patch `meta-intel-edison/meta-intel-edison-bsp/recipes-bsp/u-boot/files/0004-x86-edison-Enable-SD-card-detect-on-the-SDHCI-2.patch` was responsible for configuring the GPIO line that U-Boot uses for CD, but altering and even removing the patch produced absolutely no observable changes to the behavior.

andy-shev, 2021-10-15:
To your last paragraph: U-Bott does not have a GPIO driver for Intel Tangier, that's why changing flags do nothing. But putting 'cd-inverted' into DTS should have an effect (as we exercised in #135 for `cd_override_level`).

andy-shev, 2021-10-15:
It seems I have nailed down what's going on in U-Boot. The firmware sets the mode for SD CD pin to native (mode 1), while in Linux kernel we first switch it to mode 0 (GPIO). That's why U-Boot doesn't ignore its state.

TL;DR: please test these two patches (@htot drop anything in your tree except these two along with  https://github.com/andy-shev/u-boot/commit/20d551a84d7a, so you will have only _three_ patches on top of upstream U-Boot) https://github.com/andy-shev/u-boot/commit/87f693a59df3b76952c8f5fadf13ce586dc2b1b2 followed by https://github.com/andy-shev/u-boot/commit/c5cfab637085991ebc44b052c2790fa19048ee7d.

htot, 2021-10-16:
Sure. I see it might be worth upgrading U-Boot to 2021.10 in our image.

htot, 2021-10-20:
force pushed to htot/master  and htot/gatesgarth

humanoid2050, 2021-10-21:
Allrighty. Pulled the latest gatesgarth. Built, compiled, flashed just fine. U-Boot and kernel both react properly to the SD card being inserted and removed!

htot, 2021-10-21:
Thanks for testing @humanoid2050