edison-fw/meta-intel-edison #138: Can't use "configure_edison --setup"

mymistake, 2021-11-30:
Hello,

I flashed the Gatesgarth64 release. I was able to connect to the Edison afterwards via serial connection and saw it boot, and after that, I could log in as root.

Now I wanted to use the "configure_edison --setup" routine to generate a password, connect to Wi-Fi and allow SSH access.

However, the shell only briefly shows the prompt to set a new password, only to start filling the screen with "****". I can type meanwhile but not set a password. The only option is to abort the configuration.

Setting the Wi-Fi with "configure_edison --wifi" works. It scans for SSIDs and I can select one and set a password.

htot, 2021-12-02:
Hi, that code is here: https://github.com/edison-fw/edison-oobe/blob/master/src/configure_edison

After re-writing `oobe` we probably missed testing this. Would you have time to fix this code and send a patch?

mymistake, 2021-12-13:
I can try during the winter break if I can improve the situation.

htot, 2021-12-13:
That would be very welcome. BTW currently you should be able to press the power button to enable the builtin AP, then use your phone or laptop to connect to the web server. From there you can change device name, connect to your Wifi etc. Changing the password is not yet an option here and could use some improvement. As well as setting up a https connection instead of http.

mwallnoefer, 2022-01-06:
@mymistake I guess that I once faced exactly the same behaviour which you have described here. Would you mind retrying with the SSH access (on the new images SSH is enabled as default without any password)?

For sure the problem lies in the `_getch()` call in the `_getPassword()` function: https://github.com/edison-fw/edison-oobe/blob/6daba9fe150d0026ae8b79447bd576770cd9f965/src/configure_edison#L394. Looking at the code I guess that the serial connection is always sending control characters which makes the routine loop. On the original Edison image that had not been the case, so `_getch()` was blocking and everything worked.

mwallnoefer, 2022-02-23:
Any news here? As said we would need to address this in the respective kernel _tty_ module (or in its _terminal settings_).

mymistake, 2022-02-25:
Apologies for the delayed response - so far I have been unable to make a lot of progress. I needed to have the device working, so in the end I switched to the Debian image and just connected via connmanctl. I use the Edison as a micro-server.

htot, 2022-02-27:
connmanctl is Edison's default way to control the network, no need to install Debian for that. In fact configure_edison will perform most network changes using connmanctl.
The Debian version does have advantages of course, like a large amount of pre-built packages.

mwallnoefer, 2022-03-01:
Fix: https://github.com/edison-fw/edison-oobe/pull/16

> @mymistake I guess that I once faced exactly the same behaviour which you have described here. Would you mind retrying with the SSH access (on the new images SSH is enabled as default without any [...]

mymistake, 2022-08-16:
Please excuse the very late reply @mwallnoefer - I tried, and the issue is resolved. I was able to set a password and everything like expected.

closed 2022-08-16 by mymistake