edison-fw/meta-intel-edison #144: There is no issue with the network, but numactl cannot be downloaded.

nickfox-taterli, 2022-06-01:
I have installed VirtualBox with Ubuntu 18.04 LTS on Windows and built it, where numctl is not available, but it is confirmed that the link actually exists and the corresponding commit id exists.

I have tried many times, and tried using containers, or other Ubuntu LTS.

```
taterli@taterli-VirtualBox:/tmp$ git clone https://github.com/numactl/numactl
Cloning into 'numactl'...
remote: Enumerating objects: 1685, done.
remote: Counting objects: 100% (174/174), done.
remote: Compressing objects: 100% (79/79), done.
remote: Total 1685 (delta 102), reused 131 (delta 95), pack-reused 1511
Receiving objects: 100% (1685/1685), 1.59 MiB | 1.55 MiB/s, done.
Resolving deltas: 100% (1108/1108), done.
taterli@taterli-VirtualBox:/tmp$ cd numactl/
taterli@taterli-VirtualBox:/tmp/numactl$ git checkout dd6de072c92c892a86e18c0fd0dfa1ba57a9a05d
Note: checking out 'dd6de072c92c892a86e18c0fd0dfa1ba57a9a05d'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by performing another checkout.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -b with the checkout command again. Example:

  git checkout -b <new-branch-name>

HEAD is now at dd6de07 Release numactl 2.0.14
taterli@taterli-VirtualBox:/tmp/numactl$ 

```
```
taterli@taterli-VirtualBox:~/edison-fw/out/linux64/build$ bitbake -k edison-image
Loading cache: 100% |###############################################################################################################################################################################################################################################| Time: 0:00:00
Loaded 3725 entries from dependency cache.
Parsing recipes: 100% |#############################################################################################################################################################################################################################################| Time: 0:00:00
Parsing of 2429 .bb files complete (2428 cached, 1 parsed). 3725 targets, 341 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.50.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "i686-poky-linux"
MACHINE              = "edison"
DISTRO               = "poky-edison"
DISTRO_VERSION       = "3.3.6"
TUNE_FEATURES        = "m32 corei7"
TARGET_FPU           = ""
meta                 
meta-poky            
meta-yocto-bsp       = "hardknott:1cefabe0f0a2dd2ca8b6eb83e8e0824b72da941a"
meta-oe              
meta-python          
meta-networking      = "hardknott:786d3aa313e31a718acc9611ba3adb60573ec458"
meta-qt5             = "hardknott:43f8f539d40070a70fe89136db89bf5bb1dfe7ed"
meta-intel           = "hardknott:16bfef2d1e0d0c8c596741d0bbb820541c432f6d"
meta-intel-edison-bsp 
meta-intel-edison-distro 
meta-intel-arduino   
meta-arduino         = "master:c6036298dfc81a1bb12190da49d7589e5ceec521"
meta-acpi            = "hardknott:f9e02a3d1ecde335a4a162a99b73ea55f9689266"

Initialising tasks: 100% |##########################################################################################################################################################################################################################################| Time: 0:00:03
Sstate summary: Wanted 448 Local 17 Network 0 Missed 431 Current 1743 (3% match, 80% complete)
NOTE: Executing Tasks
WARNING: numactl-2.0.14-r0 do_fetch: Failed to fetch URL git://github.com/numactl/numactl;branch=master;protocol=ssh, attempting MIRRORS if available
ERROR: numactl-2.0.14-r0 do_fetch: Fetcher failure: Fetch command export PSEUDO_DISABLED=1; export DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"; export SSH_AGENT_PID="1320"; export SSH_AUTH_SOCK="/run/user/1000/keyring/ssh"; export PATH="/home/taterli/edison-fw/out/linux64/build/tmp/sysroots-uninative/x86_64-linux/usr/bin:/home/taterli/edison-fw/out/linux64/poky/scripts:/home/taterli/edison-fw/out/linux64/build/tmp/work/corei7-32-poky-linux/numactl/2.0.14-r0/recipe-sysroot-native/usr/bin/i686-poky-linux:/home/taterli/edison-fw/out/linux64/build/tmp/work/corei7-32-poky-linux/numactl/2.0.14-r0/recipe-sysroot/usr/bin/crossscripts:/home/taterli/edison-fw/out/linux64/build/tmp/work/corei7-32-poky-linux/numactl/2.0.14-r0/recipe-sysroot-native/usr/sbin:/home/taterli/edison-fw/out/linux64/build/tmp/work/corei7-32-poky-linux/numactl/2.0.14-r0/recipe-sysroot-native/usr/bin:/home/taterli/edison-fw/out/linux64/build/tmp/work/corei7-32-poky-linux/numactl/2.0.14-r0/recipe-sysroot-native/sbin:/home/taterli/edison-fw/out/linux64/build/tmp/work/corei7-32-poky-linux/numactl/2.0.14-r0/recipe-sysroot-native/bin:/home/taterli/edison-fw/out/linux64/poky/bitbake/bin:/home/taterli/edison-fw/out/linux64/build/tmp/hosttools"; export HOME="/home/taterli"; git -c core.fsyncobjectfiles=0 remote failed with exit code 128, output:
fatal: not a git repository (or any of the parent directories): .git

ERROR: numactl-2.0.14-r0 do_fetch: Fetcher failure for URL: 'git://github.com/numactl/numactl;branch=master;protocol=ssh'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /home/taterli/edison-fw/out/linux64/build/tmp/work/corei7-32-poky-linux/numactl/2.0.14-r0/temp/log.do_fetch.26394
ERROR: Task (/home/taterli/edison-fw/out/linux64/poky/meta/recipes-support/numactl/numactl_git.bb:do_fetch) failed with exit code '1'

```

I can see from the log that he has downloaded some content from https://mirrors.kernel.org/yocto-sources/git2_github.com.numactl.numactl.tar.gz

Full log:
https://pastebin.com/7iceq87u

htot, 2022-06-01:
I see you are using Hardknott. The numactl recipe for hardknott can be found like this: https://layers.openembedded.org/layerindex/recipe/168835/
According to that recipe it should (normally) fetch from SRC_URI = "git://github.com/numactl/numactl;branch=master;protocol=https \
But in your error message above it says protocol=ssh. Maybe you downloaded Yocto some time ago as it seems you may be missing this patch: http://cgit.openembedded.org/openembedded-core/commit/meta/recipes-support/numactl/numactl_git.bb?h=hardknott&id=ab781d4e3fa7425d96ea770ddfd0f01f62018c5b?

htot, 2023-01-25:
Were you able to resolve this?

closed 2023-11-09 by htot