edison-fw/meta-intel-edison #145: Initramfs unpacking failed: invalid magic at start of compressed archive

leshniak, 2022-06-06:
Flashing has ended successfully, but I cannot boot the system. I've tried tried with `--recovery` option. Also tried couple of releases.

Here are some logs:

`******************************
PSH KERNEL VERSION: b0182b2b
                WR: 20104000
******************************

SCU IPC: 0x800000d0  0xfffce92c

PSH miaHOB version: TNG.B0.VVBD.0000000c

microkernel built 11:24:08 Feb  5 2015

******* PSH loader *******
PCM page cache size = 192 KB
Cache Constraint = 0 Pages
Arming IPC driver ..
Adding page store pool ..
PagestoreAddr(IMR Start Address) = 0x04899000
pageStoreSize(IMR Size)          = 0x00080000

*** Ready to receive application ***

U-Boot 2021.10 (Oct 04 2021 - 15:09:26 +0000)

CPU:   Intel(R) Atom(TM) CPU  U1000  @  500MHz
DRAM:  980.6 MiB
WDT:   Started with servicing (60s timeout)
MMC:   mmc@ff3fc000: 0, mmc@ff3fa000: 1
Loading Environment from MMC... OK
In:    serial
Out:   serial
Err:   serial
Net:   No ethernet found.
Hit any key to stop autoboot:  0
Target:blank
Partitioning already done...
Flashing already done...
10083040 bytes read in 435 ms (22.1 MiB/s)
11630627 bytes read in 499 ms (22.2 MiB/s)
Valid Boot Flag
Magic signature found
Linux kernel version 5.10.63-edison-acpi-standard (oe-user@oe-host) #1 SMP Wed Sep 8 06:49:02 UTC 2021
Building boot_params at 0x00090000
Loading bzImage at address 100000 (10068704 bytes)
Initial RAM disk at linear address 0x06000000, size 25165824 bytes
Kernel command line: "quiet root=/dev/mmcblk0p8 rootfstype=ext4 console=ttyS2,115200n8 earlyprintk=ttyS2,115200n8,keep loglevel=4 systemd.unit=multi-user.target hardware_id=00"
Kernel loaded at 00100000, setup_base=00090000

Starting kernel ...

[    2.794693] Initramfs unpacking failed: invalid magic at start of compressed archive
Scanning for Btrfs filesystems
Starting version 246.9+
Kernel with acpi enabled detected
Loading acpi tables
[    3.922193] pca953x i2c-INT3491:00: failed writing register
gpioset: at least one GPIO line offset to value mapping must be specified
gpioset: at least one GPIO line offset to value mapping must be specified
Waiting for root device /dev/mmcblk0p8
Found device '/run/media/mmcblk0p8'
Init found, booting...
mount: can't find /boot in /etc/fstab
mount: mounting /boot on /realroot/boot failed: Invalid argument
mount: can't find /lib/modules in /etc/fstab
mount: mounting /lib/modules on /realroot/lib/modules failed: Invalid argument
[    8.095796] systemd[1]: Failed to start Set timezone by geolocation service.
[FAILED] Failed to start Set timezone by geolocation service.
[    9.541258] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43340-sdio for chip BCM43340/2
[    9.994526] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43340-sdio for chip BCM43340/2
[   10.044894] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   10.129412] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43340/2 wl0: Aug 29 2019 05:13:37 version 6.10.190.78 (r722178) FWID 01-e7bd1c9f
[FAILED] Failed to mount /home.
[DEPEND] Dependency failed for Post install service.
`

htot, 2022-06-06:
Looks like some new error in init script when booting ext4 image. You could try flashall with --btrfs ( works for me). Or fix the init script [here](https://github.com/edison-fw/meta-intel-edison/blob/master/meta-intel-edison-distro/recipes-core/initrdscripts/initramfs-framework/init). Or interrupt U-Boot then `run do_rescue`. From there you can mount the disk and fix /etc/fstab.

leshniak, 2022-06-07:
I've had to flash the board with the latest, official image and then I've do `flashall.sh --btrfs`, now it works! Thank you for help and your work.

closed 2022-06-07 by leshniak

htot, 2022-06-07:
Hi @leshniak thanks for testing. This means that the btrfs image is working well, but the ext4 image for some reason refuses to mount /home. I see 2 possibilities:
1) something changed in mount options (mount: mounting /boot on /realroot/boot failed: Invalid argument) causing problems in the init script.
2) there is no home partition due to move/resizing /
So that needs fixing, except I'm planning to drop ext4 in future versions. If you care to fix the issue I'll take the patch of course.

leshniak, 2022-06-08:
I've had [this image](https://www.jubilinux.org/) installed before flashing. I think it's not a common issue.