edison-fw/meta-intel-edison #158: npm i mraa fails

jessewriter, 2023-12-26:
i think that npm should work regardless of the opperating system but I am unable to find a way to get MRAA back and my code examples to work.  any suggestions to get mraa back?  or do i have to rewrite all my code using other libraries and how much code transformation is requried?

thanks!

tried :  'npm i mraa'

htot, 2023-12-26:
Hi @jessewriter I don't know exactly what `npn i mraa` does. But libmraa  and and libupm can still be built by Yocto (see recipe [here](https://git.openembedded.org/meta-openembedded/tree/meta-oe/recipes-extended/mraa/mraa_git.bb?h=master). To make mraa work with python you need the python bindings (enabled by default), and for nodejs the nodejs bindings (disabled by default), the recipe says:
```
# Note: 'nodejs' is disabled by default because the bindings
# generation currently fails with nodejs (>v7.x).
```
With mraa and nodejs bindings, I am not sure it would work correctly (haven't tried for a long time). Reason is, some pinmuxing is done by the edison layer using debugfs, which was not supported by newer kernels. However I believe @andy-shev mentioned this support may have been reinstated in the kernel. Also gpio pin configuration should now be done using libgpiod, and has been implemented in mraa for some platforms, but not edison. Specifically this file may need work to get your code working (and hasn't been touched for a long time):
https://github.com/eclipse/mraa/commits/master/src/x86/intel_edison_fab_c.c
But serial port should work and you could get pinmux working either through debufs or acpi tables and with that your code might work.

Personally, I gave up on this. We are building now like 400 iio drivers for Edison as modules for the linux kernel. Likely your devices are covered, but as nobody bothered to make a mraa translation layer your code may need significant changes.

Hard to say what needs to be done though, I have an example on a TSL2561 light sensor here with mraa :https://edison-fw.github.io/meta-intel-edison/4.5-mraa-upm.html#example-using-upm-and-mraa-on-the-tsl2561 and  using acpi / kernel / libiio here https://edison-fw.github.io/meta-intel-edison/4.6-libiio.html. That might give an idea of the work.

andy-shev, 2023-12-26:
@htot, there is official debugfs file for that for somewhile. But I **highly** recommend to forget about MRAA. Noone should use it.

htot, 2023-12-27:
Hi @andy-shev thanks for the link. I totally agree MRAA seems dead. It would have been nice to keep old code running, and it might be possible for python based code but the nodejs binding have been broken for a long while and nobody seems interested or capable to fix it.

Best is to get kernel driver working (or write kernel driver for the benefit of all mankind) and then modify user space code to use the kernel driver.

closed 2025-04-05 by htot