edison-fw/meta-intel-edison #162: Package 'libegl1-mesa' has no installation candidate

jessewriter, 2025-03-15:
Package libegl1-mesa is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

remove from setup instructions or provide a valid import

https://lists.yoctoproject.org/g/docs/topic/libegl1_mesa_not_available_in/105760426

sounds like might not be needed

htot, 2025-03-18:
I don't know. It seems I have libegl-mesa0 installed. And I can't remove it, seems whole of kde depends on it.

I seems we have a additional requirement `libegl1-mesa libsdl1.2-dev  mesa-common-dev`. However on my machine I do not have `mesa-common-dev`, but as I never run qemu-native I don't know it it's required or not.

What do you think, drop all 3 as a requirement?

htot, 2025-04-05:
Fixed in 8d7bd38b7410883a7b252ca94321f2ff55fe0fa0

closed 2025-04-05 by htot